do-gooder/rctpower_writesupport #1: Get Function showing wrong values after first calls and then the right one

heikone, 2024-08-04:
Hi, I've observed, that the call of the function `get_value` sometimes delivers strings values and after a couple of calls the expected value (here: 0.75) is delivered.

Refer to my example:
[image]

@do-gooder, can you confirm this behaviour?

do-gooder, 2024-08-04:
@heikone, do you have an app or, for example, the Home Assistant integration running?
Perhaps the RCT portal could also be causing the problem.

For me, the problem usually occurs when other requests are running at the same time.

heikone, 2024-08-04:
Good point. App might be getting values and Homeassistant custom integration [rct-power-integration](https://github.com/weltenwort/home-assistant-rct-power-integration) is running. 
Thanks, Heiko

heikone, 2024-08-04:
Confirmed. When app and integration deactivated, the values are directly as expected.

closed 2024-08-04 by do-gooder