do-gooder/rctpower_writesupport #11: No QuickCharge possible if battery.soc is smaller than power_mng.soc_min

do-gooder, 2024-10-03:
Does QuickCharge start automatically when the battery has reached power_mng.soc_min?

Is it harmful to the battery if power_mng.soc_min is set lower than battery.soc? The charging process then starts immediately. 
Observe: Is charging also throttled if a lot of solar power is suddenly generated?

heikone, 2024-10-07:
Let my try to get your question right:

For me it sounds like the question to limit the charge rate, if the battery is charged from grid.
I can remember, the charge rate can be limited by `Max. ins Last Batterie Entladeleistung [W]` in the app , but I'm not using this in my use case.
I would propose to limit to some value far below the max. charge limit - like 4000W for a RCT Power Storage 8.0 system.
But this just a guess - I'm not an electrical engineer.

Concerning the charging from the sun, I would say that the typ of inverter is limiting the charge rate. In my case, I'm using a RCT Power Storage 8.0, which limits sun charge to 8000W, I suppose.

do-gooder, 2024-12-08:
#20