do-gooder/rctpower_writesupport #12: Increasing power_mng.soc_min and power_mng.soc_charge at the same time?

do-gooder, 2024-10-13:
By default, the inverter has set a value of 0.07 (7%) for power_mng.soc_min and a value of 0.05 (5%) for power_mng.soc_charge.

When power_mng.soc_min is reached, the battery discharges through self-consumption (?) until power_mng.soc_charge is reached. The inverter then charges with the power of power_mng.soc_charge_power from the grid until power_mng.soc_min is reached.

Should the value power_mng.soc_charge also be adjusted for the “emergency power reserve” scenario so that there is really enough reserve?

do-gooder, 2024-10-13:
https://github.com/do-gooder/rctpower_writesupport/commit/a7d97751fa2c77d80e91dd59417cc10056ef1a4d

heikone, 2024-10-13:
Hi do-gooder,
I can confirm this behavior: 5% is the default low SOC threshold to start forced charging from grid until soc_min is reached again.
I guess this a meaningful reserve to prevent deep discharge for the battery to increase battery's lifetime.
In the docs of rctclient, it is called  "SOC min maintenance charge".

Your idea to set ich 2% below soc_min sounds good, but also let it stay on 5% as per default seems also okay, because self-drain will practically not be reached, if soc_min is set significantly higher than 7%!

rhoegen, 2025-01-16:
Hi all,

just an fyi: If you have an RCT Power Switch for Backup Power, be aware that setting the soc_charge value to a higher value than the SoC for Island mode will result in your system shutting down on the higher value.

E.g. if you have set the Island SoC to 0.07, SoC Min to 0.25 and SoC Charge to 0.23, you will only have backup power until the battery reaches 23%.