do-gooder/rctpower_writesupport #14: Alternative Implementation not working

philoxio, 2024-10-19:
I did the alternative implementation BUT I already had the RCT Power integration installed.
Now the entities are not showing up.
Is there a way to find out if the installation worked correctly and also find out why the entities are not showing up?
Any help is highly appreciated!

heikone, 2024-10-22:
Thanks for your report!

It looks like, that the `rctclient` is either not installed or at least can't be called.
Can you try to run the script `rct.py` in a Python environment outside of Homeassistant?
Of course, do not forget to install rctclient before (like with `pip install rctclient`).
Can you also confirm, that Weltenwort's [rct-power-integration](https://github.com/weltenwort/home-assistant-rct-power-integration.git) Homeassistant integration is properly running?

Let's try to identify your problem!

philoxio, 2024-10-23:
Thanks for you response and your efforts to help me solve this.

Somehow my Linux/Python knowledge does not seem to be enough, to get the script running outside of HASS.
But I can confirm, that Weltenwort´s HASS integration works like a charm.

heikone, 2024-10-23:
What happens, if you try to follow my implementation as described here?
[Homeassistant Integration with pyscript (Alternative 1)](https://github.com/do-gooder/rctpower_writesupport/blob/main/README.md#homeassistant-integration-with-pyscript-alternative-1)

When you test it out like this, after setting up, do you get a green mark like this:
[image]

heikone, 2024-10-23:
... it actually looks like, that the script `rct.py` is called in your setup, but the parameter is not correctly passed.
Did you consider putting the parameter in "single quotes", like in my example: `'power_mng.soc_max'`?

philoxio, 2024-11-01:
[image]

Your example, works but I still can´t see the new entities. :/

heikone, 2024-11-05:
Actually, no new entities are created!
But you can test the successful behavior with this example:

[image]

This will the set the maximum SOC to 90%.
You can observe the success after some moments in the existing entities of the RCT Power integration:

[image]

Of course, the entity `sensor.rct_power_storage_battery_state_of_charge_target` shall show 90% instead of 97% in my screenshot.

Hope that helps - otherwise please contact me again!

philoxio, 2024-11-08:
I get a green arrow when i post the "action" but the value of the entity remains unchanged. :/

heikone, 2024-11-09:
Did you also check protocoll of HA Core?
It shall report something like this:
`2024-11-09 16:55:25.430 INFO (MainThread) [custom_components.pyscript.file.rct_ha_call.rct_ha_call] rct_ha_call: Setting value 0.97 for parameter power_mng.soc_max on host 192.168.0.99`

You can extend the protocol this way:
[image]

do-gooder, 2024-12-08:
@philoxio: Is your problem solved?

heikone, 2024-12-28:
@philoxio: Any update or shall we close the issue?

closed 2025-03-12 by heikone