do-gooder/rctpower_writesupport #16: Charging battery over grid not allowed by inverter

JuriL1234, 2024-10-27:
Hello, 
i integrated your write support with alternative 2 and so far it is doing great for the battery. Thank you for that.

But for me it seems that my inverter does not allow external power to pass through to the battery.
So if i use the switch to quick charge the battery it will take everything from my PV but the way between grid and inverter is off. No power taken from the grid.

I have a PS 10.0 VKXE Software version 2.3.5678
Inside the app i can see an unchecked checkbox for "allow external power", but it is diabled. I just guess that this option is what is needed to allow the power to pass through to the battery. 

I am a user and no developer so i ask for help. Do you have an idea what i have to change to also access the correct option inside the inverter?

heikone, 2024-11-05:
Hmm, could you provide screenshots from your app?
Acutally, you need to set `Max. Netzeinspeisung aus Batterie [W]`and `Max. ins. Last Batterie Entladeleistung [W]' to value to allow transfer to and from the grid.

But of course, you need to set `power_mng.soc_strategy`to `2`before you set `power_mng.battery_power_external`to a certain charge or discharge power.

Hope that helps!

JuriL1234, 2024-11-10:
I finally found some time to invest in this. Thanks for the reply btw.
So with help of a friend i tried something and got the communication to work at least.
Getting values was not possible because rctclient was not installed via pip.
Also the automation seemed to be not up to date:
[image]
Automation was repaired like this:
[image]

Cant connect via app to my inverter... hope that works again after some time... So for now i can only provide a screenshot of my power flow chart:
[image]

For any reason it is charging with 117W. I have battery power extern of -2000. SOC min target was at 7% and i tried with 30%.
Like i said it will take all the power from the photovoltaik but nothing from the grid (or 117W whysoever).

I tried also to change p_rec_lim[1] to positive values like 200 or 2000. No effect. I dont know what "Max. ins. Last Batterie Entladeleistung [W]" is, so no change there. I also tried strategy values 0 and 1 and changed the power extern. The 117W stay whatever i do.
I am out of ideas for now...
I am still glad if you can help me with that. If it makes things easiear we can probably switch to german.
Where did you find the information for rct? I have only found the regular manual.

do-gooder, 2024-11-11:
Hi @JuriL1234,

what happens if “SOC Min Target” is set to 7% and you starting QuickCharge?

I had the same effect tonight when I tried to charge from the grid. The SOC was below “SOC Min Target”. I had the same effect as you.

I don't yet know how to deal with this.

heikone, 2024-11-13:
Can we think of logging in with service / factory access in the app and active it once?
Other way would be to set the value `power_mng.use_grid_power_enable`by using Python function `bool("True")`or `bool("False")` or any other content to the bool-function.

do-gooder, 2024-11-14:
I have added the function "power_mng.use_grid_power_enable" to rct.py.
In package the switch "Use Grid Power" is added.
The current state is not reflected!
@JuriL1234: Please update rct.py and package/rctpower.yaml

JuriL1234, 2024-11-15:
Big thanks to @do-gooder and @heikone !
"power_mng.use_grid_power_enable" does the trick. Loving it!
It is indeed the checkbox in the app i suspected to be the disturbing culprit. Although i wonder if it does any harm to always allow grid power.

Now the real pain starts to use it efficiently... you do not happen to have the perfect, easy to share strategy to use the battery together with tibbers changing market prices? :)

Btw. i found some little issues in the code:

- missing closing bracket for the state in rctpower.yaml
            data:
              object_name: "power_mng.battery_power_extern"
              object_value: "{{ states( 'input_number.rct_quickcharge_power'  | int * -1)}}"

- in rct.py the valid parameters are missing a "," for the new entry.

Great work anyway!

do-gooder, 2024-11-15:
I have correct the little issues, thanks!

I am currently working on a solution. Give me a few more days to test it before I present it here.

do-gooder, 2024-11-16:
Check packages/tibber_rct_autoload.yaml
This my test automation.

do-gooder, 2024-11-16:
Next step:
  - Find the cheapest price on your own
  - For summer: Only if the solar forecast is below XXX kWh

heikone, 2024-11-17:
Good afternoon do-gooder & JuriL1234,
I'm sorry, I'm not using Tibber or any other dynamic supplier, but maybe next year. I'm excited to so what you build.
Maybe another idea is to use the project [EMHASS](https://emhass.readthedocs.io/en/latest/#). This seems to be a very developed project.
Greetings, 
Heiko

closed 2024-12-08 by do-gooder