do-gooder/rctpower_writesupport #17: RCT commands not working properly

Buck1976, 2024-11-10:
Hi. 

Maybe it´s not a problem of the module here, but I will try to ask you all as experts. I successfully installed all modules to send commands to my PS 10.0 3LGM (Fimware 2.3.5678) connected with BMS 2.0 battery (Firmware 5608). Very often, my commands sent to the inverter via Ethernet LAN are not beeing accepted or executed. In this case I also cannot control the devices via official "RCT Power App". All items where shown "offline". After some seconds / minutes the items suddenly switch to "green" / online and after that I can successfully send commands through my writesupport scripts. In 4 of 5 cases my commands are not beeing received successfully so I scripted some workarounds to try to send the commands e.g. every minute. 

In my opinion this is caused by the RCT way of network communication. But maybe someone here has found out some "tricks" to make these scripts reliable.

Thanks and best regards.

dahadd, 2024-11-10:
Have you ever tried communication via the WLAN interface?

Buck1976, 2024-12-01:
Thanks for advise. I will try that now with my master inverter. First commands were successfullly set by this way. If this is the solution: what is the reason :-)?  I always learned that cabling is much more better than "airing". But I also see that RCT only spends 100Mbit Nics to their inverters.

Buck1976, 2024-12-02:
This night I had trouble again to charge my batterie via the module. For me it seems to be part of the problem that my 2nd inverter (slave) is not online in that moments as it´s dark outside. Only the master inverter is working at that time. If I switch during daytime (in other words:  with PV power) it seems to be more stable.