do-gooder/rctpower_writesupport #20: Tibber Autoload only 120W if Battery SOC drops under "Min SOC Zielwert"

Satyric2207, 2024-12-08:
Hi! I tested your Tibber Autoload. It basically works as expected. It starts and stops as soon as the price drops below or rises above the set value.
 
However, I have one big issue: If Min SOC goes under 7% before the start of charging it will only charge with ~120W. As soon as 7% is reached (which takes a wile) it jumps up to the value set in rct_quickcharge_power - in my case 8000W.

I learned that as soon one is changing "Min SOC Zielwert" to e.g. 0.07 the value "SOC min Erhaltungsladung" sets itself to 0.05. Over the night it drops slowly by the power usage of the inverter near 5% and charges up to ~7% with around 120W. 

I helped myself by charging the battery up to a value which guarantees me SOC over 7% late at night/early in the morning. It works but is unhandy. Second idea was to set the "Min SOC Zielwert" to 9% and as a result "SOC min Erhaltungsaldung" is now 7%. But unfortunately (and as expected) it didn't work out that easily. 3rd idea was to reduce the value "Min SOC Zielwert" by 2% (-0.02) which works fine.

To make this a little more sophisticated I looked for the value "SOC min Erhaltungsaldung" in [RCT Power ](https://github.com/weltenwort/home-assistant-rct-power-integration)for a proper calculation, but I didn't find any 😕.

I don't know if anybody has the same problem, yet.
Maybe you have a good solution for this?

dahadd, 2024-12-08:
Look at alod APP Version 2.591 f.e. APTOIDE
There is an option for the trickle charge.

do-gooder, 2024-12-08:
`power_mng.soc_charge_power` can be set via `rct.py`. The default value is 100 (W). Change it to speed up the charging to "Min SOC Zielwert".

If you have developed a solution, please send us a pull request!

do-gooder, 2024-12-08:
Shouldn't this deal with exactly your topic?
Line 44-48 of `packages/tibber_rct_autoload.yaml`
```
- action: shell_command.writerctpower
  data:
    object_name: power_mng.soc_min
    object_value: >-
      {{states('input_number.rct_soc_min')|multiply(0.01)-0.02|round(2)}}
```

do-gooder, 2024-12-08:
@Satyric2207, check [Lowering of min_soc increased](https://github.com/do-gooder/rctpower_writesupport/commit/5557a1cc0dae011a067e2aaf850105ee02f79a1c)

Satyric2207, 2024-12-23:
@do-gooder
Thanks for your the quick feedback and sry for my late. I think the solution for me is to change the trickle charge value as @dahadd wrote.

@dahadd 
Can you please tell us what values you are using and if you have discovered any issues with them? Thanks in advance!

closed 2024-12-23 by Satyric2207