do-gooder/rctpower_writesupport #23: Battery to grid

puhycz, 2025-01-17:
Is it possible to issue a command to discharge the battery back to the grid?

Thanks.

TobiasHuber1980, 2025-01-17:
By default, feeding power from a battery into the grid is disabled.

To enable this, change the following parameter (see the RCT Power App):

Device -> Settings -> Battery -> Max. battery to grid power [W]

The corresponding parameter is:

0x54829753 // p_rec_lim[1] | t_float | Max. battery to grid power [W]

This should be a positive value, typically equal to the inverter's maximum AC power (e.g., 10000).

puhycz, 2025-01-17:
Many thanx for your response. Is save for inventer FLASH memory? How often can I change parameters of SOC etc …

TobiasHuber1980, 2025-01-18:
I think no problems with flash/ram. It's like changing values minsoc etc....

puhycz, 2025-01-20:
Thanks. I use HA in docker on Synology NAS. Folder of RCT integrations looks like this. 

[image] Please how install your [rctpower_writesupport](https://github.com/do-gooder/rctpower_writesupport) ? Thanks.

heikone, 2025-03-12:
Hi, I'm not using HA in Docker of Synology, but it shall be work as described in our alternative "installations":

1. [as PyScript](https://github.com/do-gooder/rctpower_writesupport?tab=readme-ov-file#homeassistant-integration-with-pyscript-alternative-1)
2. [as package](https://github.com/do-gooder/rctpower_writesupport?tab=readme-ov-file#home-assistant-integration-as-package-alternative-2)

Greetings, Heiko

closed 2025-03-12 by heikone